functionreturnfunction/G27_Pedals_and_Shifter #1: G27 Pedals with a loadcell

pedromelim, 2016-05-30:
Went ahead and used all of this that you made available (btw tks alot, really nice you made all of this available to everyone), but im having an issue.

I have my G27 pedals with a loadcell on the brake, and i think thats why the brake pedal doesnt work with this sketch. Im guessing it must be because the loadcell functions in a different way than a potentiometer, its like the axis is inverted on the brake pedal. Trying to see the sketch for a fix but would be nice to have some help from the person that build this in the 1st place, like that could be implemented to other people around.

functionreturnfunction, 2016-05-31:
Most sims I've seen have an option to invert axis for your pedals...but I needed something to do while I had my coffee and you asked so very nicely.  Current master, as of dfe1f6f, you'll want to uncomment line 25 before building and loading.

Now, what load cell are you using?  Is it a kit like APElectrix/RicmoTech, or something you built yourself?

pedromelim, 2016-05-31:
Its a custom load cell, was cheaper to do hehe.

Pedal wasnt even working on the game controllers, thats why i found it weird.

Need to get home to do that. I will report to you once i have done testing.

FYI: Load cell mod was about 50/60€ so if you wanna do it yourself, i can try and help you out on doing it...no sodering skills needed, did all of it without sodering at all.

functionreturnfunction, 2016-05-31:
Sure, I would absolutely appreciate any info you'd be willing to share, I've wanted a load cell for a while now but the prices have been holding me back.  Even just a parts list, where you got them from, and maybe some pictures would be great.

pedromelim, 2016-05-31:
just tried it and doesnt work...uncomented the debug line to see the serial monitor stuff, and it seems the max and min value are always around 500, when i press the pedal almost nothing changes, its weird...

functionreturnfunction, 2016-05-31:
Hmm, try uncommenting line 28 also (STATIC_THRESHOLDS).  The default is a self-calibrating mode where it determines the min and max possible values for each pedal based on the min and max values the device has seen since it was last restarted, so you have to push each of them all the way in to calibrate it.

Those thresholds are specifically for my own pedals, I've done the [quiet mod](http://www.isrtv.com/forums/topic/10602-quieter-pedal-mod-for-g25g27/) to my set so they have a little less travel than stock.

pedromelim, 2016-05-31:
Done it, didnt change...been doing some tests and as it is now, it seems im losing a bit of voltage from the arduino to the pedals...all i get is close to 3v when im suposed to be having 4v at least....measured the output from the pedal at it reamins always at abut 1.5v on the brake...the other ones are fine, and go up and down as i press them and release....the brake pedal remains at same voltage wich is very weird....will do some tests end of the week or weekend perhaps...gonna a couple of busy days ahead.

functionreturnfunction, 2016-05-31:
That's strange.  Does the load cell use/require any kind of power supply?

I had weird results when I tried to add an LED display module to the design, the pedals would only go up to about 50% when I pressed them all the way down, and the solution was to use VCC rather than RAW as supply voltage for the display (and add an external power supply).

pedromelim, 2016-10-29:
Ok, so after some months without making anything regards to this, i went ahead and did some work today.
I figured why it didnt work, somehow after connecting the pedals cable to the arduino, where it was supposed to have positive voltage, it was negative...so when i reached this, i switched negative to positive (on the loadcell only) and it worked!!! After this i took a look and the inverted function you did works as well, my question is, do you know why this negative voltage happens? and if there is someway we could make it work like it would work with the pedals connected to the G27?! Anyway, at least now i do know it Works ;)