functionreturnfunction/G27_Pedals_and_Shifter #2: Schematic for soldering

MathijsG, 2016-08-23:
Hi,

Where can I find a graphical schematic for soldering this? I really like this project. I have a G27 wheel set without the wheel, and I don't think I will get the wheel easily so I want to use this project to hook the shifter and pedals to my pc, while using my old MOMO wheel.

Does the clutch also still function like it does with the G27? For shifting?

RIP95, 2016-08-25:
You don't need scheme to do it. Just buy any ATMega32u4 with Arduino Leonardo bootloader, get few wires, place and solder them according to pinout table provided in sketch, connect to ports (or you can cut connectors and solder them directly to board, if you don't have any warranty and don't afraid to solder them back), adjust some values if you need. And congrats, you got shifter and pedals working without base.
That's how it works for me.
[image]

MathijsG, 2016-08-25:
Nice. I saw a shitload of resistors on the main page. That's why :-). I will try the schematic once I have the connector.

closed 2016-09-29 by functionreturnfunction

erkan377, 2022-02-08:
[image]
[image]