functionreturnfunction/G27_Pedals_and_Shifter #3: 'G27' was not declared in this scope.

MathijsG, 2016-08-26:
I get this error while uploading the sketch to my Arduino.

```
"Arduino: 1.6.9 (Windows 7), Board: "Arduino Nano, ATmega328"

E:\Users\Mathijs Groothuis\Documents\Arduino\G27_Pedals_and_Shifter-master\G27_Pedals_and_Shifter\G27_Pedals_and_Shifter.ino: In function 'void setXAxis(void*)':

G27_Pedals_and_Shifter:180: error: 'G27' was not declared in this scope

   G27.setXAxis(input->axis);

   ^

E:\Users\Mathijs Groothuis\Documents\Arduino\G27_Pedals_and_Shifter-master\G27_Pedals_and_Shifter\G27_Pedals_and_Shifter.ino: In function 'void setYAxis(void*)':

G27_Pedals_and_Shifter:185: error: 'G27' was not declared in this scope

   G27.setYAxis(input->axis);

   ^

E:\Users\Mathijs Groothuis\Documents\Arduino\G27_Pedals_and_Shifter-master\G27_Pedals_and_Shifter\G27_Pedals_and_Shifter.ino: In function 'void setZAxis(void*)':

G27_Pedals_and_Shifter:190: error: 'G27' was not declared in this scope

   G27.setZAxis(input->axis);

   ^

E:\Users\Mathijs Groothuis\Documents\Arduino\G27_Pedals_and_Shifter-master\G27_Pedals_and_Shifter\G27_Pedals_and_Shifter.ino: In function 'void setButtonStates(int*, int)':

G27_Pedals_and_Shifter:286: error: 'G27' was not declared in this scope

     G27.setButton(i, LOW);

     ^

G27_Pedals_and_Shifter:290: error: 'G27' was not declared in this scope

     G27.setButton(gear - 1, HIGH);

     ^

G27_Pedals_and_Shifter:294: error: 'G27' was not declared in this scope

     G27.setButton(buttonTable[i], buttons[i]);

     ^

E:\Users\Mathijs Groothuis\Documents\Arduino\G27_Pedals_and_Shifter-master\G27_Pedals_and_Shifter\G27_Pedals_and_Shifter.ino: In function 'void setup()':

G27_Pedals_and_Shifter:355: error: 'G27' was not declared in this scope

   G27.begin(false);

```
   ^

exit status 1
'G27' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
"

RIP95, 2016-08-27:
As I can see from a report you are using Arduino Nano, I'm not sure if you can do the same thing with it. G27 class isn't declared, because of "#if defined(_USING_HID)" and that's why you are getting this error. The error cause because of your board HID library which probably doesn't support such thing.

MathijsG, 2016-08-27:
Yeah I guessed that was the problem, because when I put the board on 'Sparkfun Pro Micro' I didn't get any compiling errors. I bought a u4 board from Aliexpress, hope to play with this soon :-).

MathijsG, 2016-09-28:
I now have this u4 board and flashed the sketch. But what kind of device needs it to be? Because Euro truck simulator doesn't recognize anything. Could you clarify the project some more?

functionreturnfunction, 2016-09-29:
Start -> Run -> type "joy.cpl" and press enter.  Your Android Leonardo should show up in the listing, click that and click Properties.  This should open a window that will respond to anything you do with the shifter or pedals, if it doesn't show up in the listing at all then it may not have flashed properly.

MathijsG, 2016-09-30:
Unfortunately no such apparatus does show up here. When I load up other sample code of emulating mouse and keyboard they do show up as keyboard. What kind of device should I see? A joystick named G27?

MathijsG, 2016-09-30:
Somehow downloaded it again and now a separate virtual thing showed up. It's working, NICE! :-)

closed 2016-09-30 by MathijsG