functionreturnfunction/G27_Pedals_and_Shifter #4: Questions about the led system

MathijsG, 2016-10-04:
Now I have a working unit, many thanks, I am trying to get the leds to work. I don't have an RGB led, so I just use seperate leds per input 3, 5 and 6, and connect them all thee to ground. However I don't see anything light up. Is this led functionality something that inherently needs to work when using the shifter and pedals, or must it be supported by the game?

Can you clarify this for me? Thanks :)

functionreturnfunction, 2016-10-04:
The LED is completely unnecessary, I only used one originally to debug pedal inputs before I added the USB controller library.  I've had a big bag of those RGBs for years and this was the first opportunity I've run into to use any of them.

MathijsG, 2016-10-04:
I already got it working after uncommenting. Although the leds are also somewhat lighting up while not pressing the pedals, maybe because of the lack of a pull-up resistor.

closed 2016-10-11 by MathijsG