functionreturnfunction/G27_Pedals_and_Shifter #5: Games hard to bind to keys, in Windows it's working perfectly.

MathijsG, 2016-10-11:
Don't know if it's just me, but in the Windows dialog, every key and axis is working perfectly, but when I want to assign functionality to a certain button (for example in Euro Truck Simulator) I get a hard time registering the button. But when I look in the Windows dialog, the button IS pressed.

Is it just me, or is it quite hard sometimes?

functionreturnfunction, 2016-10-26:
I haven't had any problems as such, I don't have Euro Truck Simulator though.

ming982566, 2017-11-09:
Well, yes.
I also have problems like MathijsG, in the game of Euro Truck Simulator.
I'm thinking, if only identify the push rod in place of the button, without identification in the middle of the process, (is the output shaft that game controllers are removed, direct output button) that could solve the problem, but I don't know how to modify the code.

lshachar, 2019-07-03:
sorry; this is extremely old I know - I found that if I had more than one arduino as joystick, since they get the same VID and PID, windows / the game has issues knowing which device is which. my solution was to add entries to boards.txt so that I could rename some arduino pro (leonardo) devices, and give them different VID / PID.

MathijsG, 2019-07-04:
In my case I used a single Arduino for the shifter and the pedals, so that couldn't work for me. However I own a G27 for years now, so it's not an active issue for me anymore.