functionreturnfunction/G27_Pedals_and_Shifter #6: Value to fill in at pedal axis treshold when manually defining the range

MathijsG, 2016-10-26:
What pedal values need to be filled in when you don't want the calibration all the time and just setting the values one time?

I had these values:

#define MIN_GAS     10
#define MAX_GAS     1015
#define MIN_BRAKE   5
#define MAX_BRAKE   1015
#define MIN_CLUTCH  12
#define MAX_CLUTCH  1020

It worked, but I'm not sure if these are the good values. When I go into serial mode, to debug the values I see different confusing values.

For example:

PIN: 18 GAS: 899 MIN: 38 MAX: 925 X VALUE: 993
PIN: 19 BRAKE: 155 MIN: 45 MAX: 811 Y VALUE: 146
PIN: 20 CLUTCH: 122 MIN: 11 MAX: 894 Z VALUE: 128

What do I need to fill in as the max when currently pressed to the max? Just the max value? And why differs the value of "VALUE:" and the value right next to "GAS:" "BRAKE:" "CLUTCH:"

Can you elaborate on these values? In the code I see that the value behind "GAS: " is 'input->cur' and the value behind VALUE:  is 'input->axis'. But why they differ I don't understand.

functionreturnfunction, 2016-10-26:
To enable those constants for the pedal thresholds and turn off automatic calibration mode you need to uncomment the line #define STATIC_THRESHOLDS true.

The values you see when debugging for 'MIN' and 'MAX' are the min and max values that have been read for a given pedal since the program has first started.  If you wanted to determine good threshold values, run with the serial console connected and pedal debugging enabled, press each of the 3 pedals down as hard as you can, and then choose values that are close to the min and max for each pedal.

For your second question about the pedal structs (GAS vs VALUE and cur vs axis), that's done so that the pedals can each send a value between 0 and 1023.  The AVR board is able to pull a 10 bit value, but because of the design of the pedals the potentiometers are never fully open or fully closed.  So the values are scaled from whatever the min/max physically possible are to 0 - 1023 in the function axisValue.

MathijsG, 2016-10-30:
Hi, while doing this manual static tresholds, and also with the automatic calibration I still get the problem that it doesn't scale to the whole end. I cannot reach 1023, and in some games I thus can't reach to the full extent of the axis. Is this because in reality the range is less than 0 - 1023 so you can't really scale to those values? Or should I decrease: 
#define MAX_AXIS            1023

functionreturnfunction, 2016-10-31:
Using this example:

```
PIN: 18 GAS: 899 MIN: 38 MAX: 925 X VALUE: 993
PIN: 19 BRAKE: 155 MIN: 45 MAX: 811 Y VALUE: 146
PIN: 20 CLUTCH: 122 MIN: 11 MAX: 894 Z VALUE: 128
```

I would use the following values:

```
#define MIN_GAS     39
#define MAX_GAS     924
#define MIN_BRAKE   46
#define MAX_BRAKE   810
#define MIN_CLUTCH  12
#define MAX_CLUTCH  893
```

Note that the MIN values are all 1 more than the minimums that were read for the pedal, and the MAX values are all 1 less than the maximums.

The pedals themselves will never read the full 1023 (or 0) because the potentiometers physically cannot reach those extents.  The code scales the current value to 0-1023 based on the MAX and MIN thresholds.

MathijsG, 2016-10-31:
I will try the +1 difference, I just filled in the exact same size. Will report back. Thanks for your help :-).

MathijsG, 2016-11-01:
With these values the limits are spot on! Thanks. Now I can reach to the full 1023!

closed 2016-11-01 by MathijsG