functionreturnfunction/G27_Pedals_and_Shifter #7: Option to calibrate the shifter values in a procedure

MathijsG, 2016-10-26:
Because every shifter is different in settings, maybe add a procedure for calibrating the positions? Just like you do with the axes. I used the serial monitor for manually checking the correct values. But I don't know if any user of the software is that handy?

My values are:
// SHIFTER AXIS THRESHOLDS
#define SHIFTER_XAXIS_12        330 //Gears 1,2
#define SHIFTER_XAXIS_56        600 //Gears 5,6, R
#define SHIFTER_YAXIS_135       750 //Gears 1,3,5
#define SHIFTER_YAXIS_246       200 //Gears 2,4,6, R

With your new values, I still get the problem I also got with the previous settings: in the 1st and 2nd gear with just a little wiggle the 3th and 4th gear get touched already. With the first value to 330 I got it okay.

functionreturnfunction, 2016-10-26:
Unfortunately those axis values will vary greatly between shifters.  I had to take mine apart to fix reverse gear recently and it changed a bit, which is why those values were updated recently.

To fix your own axis values, uncomment #define DEBUG_SHIFTER, upload the sketch to your device, and open the serial console.  SHIFTER_XAXIS_12 should be low enough so that you can put the shifter in 3rd and push left on it without going into 1st, and put the shifter in 4th and push left on it without going into 2nd.  Likewise, SHIFTER_XAXIS_56 should be low enough so that pushing to the left from 5th doesn't catch 3rd, and pushing to the left from 6th doesn't catch 4th.

I had thought about making some sort of user calibration app, but I've  been in the process of replacing my G27 lately.

n-e-y-s, 2020-06-15:
Maybe you want to have a look at my fork for this: 

https://github.com/n-e-y-s/G27_Pedals_and_Shifter/tree/n-e-y-s_devel

MathijsG, 2020-06-15:
> Maybe you want to have a look at my fork for this:
> 
> https://github.com/n-e-y-s/G27_Pedals_and_Shifter/tree/n-e-y-s_devel

Nice work! Although I found an original steering wheel for the G29 in the end, so I sold my altoid box and switched to G27 native usage, I like your work :).