functionreturnfunction/G27_Pedals_and_Shifter #8: Shifter button problems

pedromelim, 2016-11-01:
Shifter buttons from 8 to 19 are always pressed and i cant find where the issue is, also when using 6th and 7th gear, it will always do 7th gear. All conections are done but somehow thsi doesnt worked. I tried looking at the code for several hours but nothing found, could you help me out?
[image]

pedromelim, 2016-11-02:
After alot of time lost, just realized that the pins 0 and 1 on the arduino are switched, tought it start 0, 1, 2 etc, but its 1, 0, 2 etc...im not at home but im sure thats the problem. feel such a noob ;)

functionreturnfunction, 2016-11-02:
Sorry I didn't get back sooner, but yeah, those buttons are wired on a matrix so the fact that you see any of them triggering at all makes me think you must have a pin switched or a short somewhere.  Good luck!

closed 2016-11-02 by functionreturnfunction