functionreturnfunction/G27_Pedals_and_Shifter #9: Can't compile

Endprodukt, 2016-11-10:
Thanks for your work!

I'm on 1.6.12 - have a clone board that looks exactly like yours (32u4,  5v, 16mhz). 
I downloader the SparkFun Board library, selected the correct board (Pro Micro) - getting this every time 

Build-Optionen wurden verändert, alles wird neu kompiliert
C:\Users\OLIVER~1\AppData\Local\Temp\ccDc3yAO.ltrans0.ltrans.o: In function `main':

ccDc3yAO.ltrans0.o:(.text.startup+0x15a): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x15c): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x15e): undefined reference to `G27_::begin(bool)'

ccDc3yAO.ltrans0.o:(.text.startup+0x24c): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x24e): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x250): undefined reference to `G27_::setXAxis(unsigned int)'

ccDc3yAO.ltrans0.o:(.text.startup+0x260): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x262): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x264): undefined reference to `G27_::setYAxis(unsigned int)'

ccDc3yAO.ltrans0.o:(.text.startup+0x274): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x276): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x278): undefined reference to `G27_::setZAxis(unsigned int)'

ccDc3yAO.ltrans0.o:(.text.startup+0x378): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x37a): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x37c): undefined reference to `G27_::setButton(unsigned char, unsigned char)'

ccDc3yAO.ltrans0.o:(.text.startup+0x394): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x396): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x398): undefined reference to `G27_::setButton(unsigned char, unsigned char)'

ccDc3yAO.ltrans0.o:(.text.startup+0x3bc): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x3be): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x3c0): undefined reference to `G27_::setButton(unsigned char, unsigned char)'

ccDc3yAO.ltrans0.o:(.text.startup+0x3ca): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x3cc): undefined reference to `G27'

ccDc3yAO.ltrans0.o:(.text.startup+0x3ce): undefined reference to `G27_::sendState()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board SparkFun Pro Micro.

This is the board I'm using: 

http://www.ebay.de/itm/301951688967?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

functionreturnfunction, 2016-11-11:
That appears to be the correct board, have you tried flashing a different sketch to it?

Endprodukt, 2016-11-11:
Just tried and wrote the "KeyboardAndMousControl" to it. Worked right away. Do I have to use a USBASP to get it back to stock now?

Endprodukt, 2016-11-12:
Tried a few other sketches that came with arduino ide and they all work. Since I can't compile your sketch it seems the board itself is not the problem. 

The <hid.h> is marked red, could that be a problem?

Endprodukt, 2016-11-13:
Got it working with 1.6.9

closed 2016-11-13 by Endprodukt