functionreturnfunction/G27_Pedals_and_Shifter #11: Tracing key press.

fmit, 2017-01-21:
Hey.  Thanks for this code and instructions, shifter is working just fine.

However, I can't seem to get the buttons on G27 shifter to register at all.  the gears are fine, buttons 1-6 in Windows, but even when in debug mode, no button presses are registered.

Which Pin should I focus on to see if I have an issue somewhere?  Or even what section of the code detects the button presses?

Thank you in advance!

functionreturnfunction, 2017-01-23:
The shifter pins are laid out starting here: https://github.com/functionreturnfunction/G27_Pedals_and_Shifter/blob/master/G27_Pedals_and_Shifter.ino#L54

Specifically you're going to want to check pins 0, 1, 4, 7, and 9, you may have a couple switched around or a bad solder joint.  The button presses are calculated in getButtonStates(int) at line 227.

fmit, 2017-01-23:
Thanks for the reply.  I'll check it out and report back what my specific issue was.

fmit, 2017-01-24:
Ok.  Resoldered everything and put on a new DB9 connecter and still the same issue.  So I started tracing the cables and making sure all was good.  Could not find any issue.

Then, just out of sheer "lets try it and see" I swapped Pin 1 and Pin 2 around on the DB9 / Pro Micro.

In the code, it says this:

// SHIFTER PINS
//| DB9 | Original | Harness | Shifter | Description             | Pro Micro   
//|   1    | Purple   | Purple    |         1 | Button Clock           | pin 0       
//|   2    | Grey      | Blue       |         7 | Button Data             | pin 1       

I found I had to put DB9 1 to 1 on the Pro Micro and DB9 2 to 0 on the Pro Micro.

Getting this:

//| DB9 | Original | Harness | Shifter | Description             | Pro Micro   
//|   1    | Purple   | Purple    |         1 | Button Clock           | pin 1       
//|   2    | Grey      | Blue       |         7 | Button Data             | pin 0    

Now all buttons and gears work fine.  
My Pro Micro is a "china copy" so it might be peculiar to my board.

Anyway - That is what fixed it for me.   Thanks for the help.

closed 2017-01-30 by functionreturnfunction

tirol3, 2021-01-28:
hi thanks for this codes. can we add extra buton this project ? can you help?