functionreturnfunction/G27_Pedals_and_Shifter #12: G920 pedals

dazza83, 2017-06-04:
Hey, just wondering if this will automatically work with the g920 pedals? if not is it a easy fix?

functionreturnfunction, 2017-06-04:
Hard to say.  I'm pretty sure the g25 and g27 pedals were identical, and there's most likely no reason for the pinouts to have changed, but that doesn't mean they haven't.  I would build the adapter on a breadboard first in order to test, and switch things around as necessary if it doesn't work.  If you can find a pinout diagram for the g29 pedals that would be a big help, I just poked around google images a bit and wasn't able to.

dazza83, 2017-06-04:
Ok pretty sure its a identical pedal set. i'll give it a try tomorrow.

Also im interested in connecting up a small button box. Is it possible to this board?

functionreturnfunction, 2018-05-16:
Not just the files, you'd also need a SparkFun Pro Micro, along with some DB9 connectors wired to it.  But yes, this should work with just the pedals without the shifter if that's your goal.  I can't speak to whether or not this would work with hall sensors as I haven't used them, but odds are they would work with little modification if they act as a 10k variable resistor like the potentiometers in the pedals.