functionreturnfunction/G27_Pedals_and_Shifter #13: Ask for instructions on how to solder the connector to arduino

Lisu25, 2017-07-01:
Hello, can you send me how to solder the db9 connector to arduino micro?

MathijsG, 2017-07-03:
See the file G27_Pedals_and_Shifter.ino:

`// PEDAL PINS
//| DB9 | Original | Harness | Description | Pro Micro   |
//|   1 | Black    | Red     | +5v         | +5v         |
//|   2 | Orange   | Yellow  | Throttle    | pin 18 (A0) |
//|   3 | White    | White   | Brake       | pin 19 (A1) |
//|   4 | Green    | Green   | Clutch      | pin 20 (A2) |
//|   5 |          |         |             |             |
//|   6 | Red      | Black   | GND         | GND         |
//|   7 |          |         |             |             |
//|   8 |          |         |             |             |
//|   9 | Red      | Black   | GND         | GND         |`

`// SHIFTER PINS
//| DB9 | Original | Harness | Shifter | Description             | Pro Micro   |
//|   1 | Purple   | Purple  |       1 | Button Clock            | pin 0       |
//|   2 | Grey     | Blue    |       7 | Button Data             | pin 1       |
//|   3 | Yellow   | Yellow  |       5 | Button !CS & !PL (Mode) | pin 4       |
//|   4 | Orange   | Orange  |       3 | Shifter X axis          | pin 8  (A8) |
//|   5 | White    | White   |       2 | SPI input               |             |
//|   6 | Black    | Black   |       8 | GND                     | GND         |
//|   7 | Red      | Red     |       6 | +5V                     | VCC         |
//|   8 | Green    | Green   |       4 | Shifter Y axis          | pin 9 (A9)  |
//|   9 | Red      | Red     |       1 | +5V                     | VCC         |`

And see here the pinouts for female and male:
[image]
[image]

Lisu25, 2017-07-03:
Thank you very much :)

Lisu25, 2017-07-03:
one more question. Where is "spi input"?

MathijsG, 2017-07-03:
As you can see in the code table of  G27_Pedals_and_Shifter.ino there is no pin location given, thus no pin needed.

Look at the original tables in the file:

[image]

As there are other pins which do not need to be connected.