functionreturnfunction/G27_Pedals_and_Shifter #35: Shifter button problems.

Emix1983, 2024-11-07:
Since I came from a different project, I have different connections compared to the current one. Specifically, the DB9 pins 1 and 2 are connected to pins 7 and 5 of the Arduino. I have defined the variables accordingly to reflect these changes. However, aside from the shift for 6 gears, none of the other buttons are functioning. I'm starting to wonder if this issue is related to the differences in these pin connections. Could anyone offer some advice? Feel free to let me know if you need further assistance!

YepCaptain, 2024-11-07:
I guess you have a G25 shifter. It is different from the G27, so you should swap the 1 and 7 wires on the db9 connector and then the buttons will work. I ran into the same problem today and it was not easy to find a solution. 
[image]

Emix1983, 2024-11-08:
>  I guess you have a G25 shifter. It is different from the G27, so you should swap the 1 and 7 wires on the db9 connector and then the buttons will work. I ran into the same problem today and it was not easy to find a solution. 
> [image]
> 
Thanks sir.

I don't think because once a upon a lot of time ago 😆 I bought and have a G27. But this afternoon I will try to swap seeing your reply!